nagimov/cyclotron-cryopanel #1: create a simple case for heat exchanger with 2 flows

nagimov, 2016-09-03:
We will need two cases - parallel flow and counter flow.
Then we're going to extend it for 4 flows (nitrogen + helium).
First iteration could have been done without refprop integration.

nagimov, 2016-09-06:
initial revision committed

nagimov, 2016-09-07:
TODO:

- [ ] fix variable names according to wiki
- [ ] fix long formulations (split to sub-variables)
- [ ] simplify for constant `UA`
- [ ] fix result plots according to required results

savreline, 2016-09-07:
New plots, including new one of temperature vs. length of heat exchanger. 
[image]
[image]

savreline, 2016-09-07:
To do list is done, code updated on github
ode45 matrix described in the MD file

savreline, 2016-09-07:
New plots after missing Cp added
[image]
[image]

closed 2016-09-07 by nagimov