nagimov/cyclotron-cryopanel #2: create a simple case for heat exchanger with 4 flows

nagimov, 2016-09-04:
Extend two-flows case for 4 flows

nagimov, 2016-09-07:
to be done after finishing 2-flow configuration

nagimov, 2016-09-08:
to be reopened

savreline, 2016-10-24:
@nagimov Got the code to solve both sides of the cryopanel. One side of the cryopanel is 200-150-100 (N2, wall, He) and the other side is 100-75-50. This is done to distinguish between the two in the testing stages. Will link the streams later on.

[image]
[image]
[image]

Will now work on adding more features (heat conduction between streams)

nagimov, 2016-10-31:
done

closed 2016-10-31 by nagimov