nagimov/cyclotron-cryopanel #5: Get Cp from refprop as a function of T

savreline, 2016-09-07:
Issue 1, remove assumption that `Cp` is constant. 

Get `Cp` from refprop `f(T)`

savreline, 2016-09-07:
Code uploaded. Refprop works but reaches limit for nitrogen at below 63K due to limitations in the equation of state. Help says need to edit refprop file and pick a different equation of state. Not really and issue right now, will just keep the program above 63K.

savreline, 2016-09-07:
In refprop quality is set to 0, as we are dealing with one-phase liquid streams for now. This becomes a problem if He is above critical point (5.3K). Will address later, so far will just stay below 5.3K

savreline, 2016-09-07:
Figures 
[image]
[image]

nagimov, 2016-09-07:
branch renamed to `issue_05`

savreline, 2016-09-07:
Ah yeah my mistake, thanks!

nagimov, 2016-09-07:
@savreline also let's do some housekeeping on our images here.
After you're done with uploading your image, convert the original image's code to html, so it looks tidier.
E.g.
original `![plot_t_vs_length](https://cloud.githubusercontent.com/assets/21995425/18329641/6265f09a-7509-11e6-9cff-f97d27eac7d6.png)` will become `<img src="https://cloud.githubusercontent.com/assets/21995425/18329641/6265f09a-7509-11e6-9cff-f97d27eac7d6.png" width=150px>`

nagimov, 2016-09-07:
@savreline just a nitpick - let's for future revision specify mass flow (not volumetric flow)

savreline, 2016-09-07:
Code is updated using `mid` and `last` for indices which makes indices much easier to read and figure out what is going on.

savreline, 2016-09-07:
@nagimov Changing to mass flow would be more *natural* once we get enthalpy in, just my opinion

savreline, 2016-09-07:
[Link](http://www.boulder.nist.gov/div838/theory/refprop/FAQ2.htm) to discussion about equation of state limits in refprop and how to modify them

nagimov, 2016-09-07:
@savreline that's a dangerous zone, let's not do that. If we go below the limits, something is wrong in either initial state or the code. For now just keep things away from the limits

savreline, 2016-09-07:
The limit is 63K for nitrogen... so as long as nitrogen doesn't cool below 63K in the cyclotron in real life, we are good.

nagimov, 2016-09-07:
@savreline Temperature in the LN2 circuit of the cyclotron doesn't go below 77.4K. Added to a meeting todo for discussion.

closed 2016-09-07 by nagimov