nagimov/cyclotron-cryopanel #8: Removing the one-phase assumption

savreline, 2016-09-07:
Introducing equations in terms of quality, enthalpy and internal energy.

savreline, 2016-09-07:
This will probably require some research since I'm not sure if `ode45` would work here. Need to examine in detail.

savreline, 2016-09-07:
Yeah we no longer have an ***explicit*** problem. We have an ***implicit*** problem, so we have to use some ***implicit solver*** like `ode15i`. Could take a while to set up and got to hope for the best it would work!

savreline, 2016-09-08:
@nagimov Tomorrow morning I will work on setting up 'ode15i' and see how it goes. If it works and you like it, we can keep it. If not, we can get your method working. Or get it working regardless, but I would like to try `ode15i` first.

savreline, 2016-09-08:
@nagimov I don't think the `ode15i` method would work here, because `dT/dt` is not even present. I'm setting up your finite difference method right now. When I finish the finite differences, if I get new ideas about `ode15i` I will try it again.

savreline, 2016-09-08:
First version of code uploaded. Contains constant HX_UA assumption, but that should be easy to remove.

nagimov, 2016-09-08:
Awesome. Let's test the consistency of this solution with ode45

savreline, 2016-09-08:
Assumption of constant HX_UA is removed, some minor housekeeping done. But the numbers don't look that good (didn't look that good either when HX_UA was constant). Perhaps we should check to see if the initial data makes sense, etc.

savreline, 2016-09-08:
Essentially `fsolve` in MATLAB is an iterative procedure and produces a bad solution. The iterative method doesn't seem to wanna do much work - produces a solution not far away from the initial guess. Or doesn't find a solution. Somehow got to make it put more effort into looking for a solution.

savreline, 2016-09-08:
The most realistic solution so far 

[image]

savreline, 2016-09-09:
Current ideas on how to proceed are summarized in the LaTeX document 

[fsolve_issues_Sept9.pdf](https://github.com/nagimov/cyclotron-cryopanel/files/464501/fsolve_issues_Sept9.pdf)

savreline, 2016-09-09:
Added list of relevant MATLAB solvers to the .md file. 

So far tried `fminsearch`, code works, but hangs up on refprop going into the wrong area.

savreline, 2016-09-09:
Will study different options for `fsolve` and there is a choice of solvers within `fsolve` to use, will go through those.

savreline, 2016-09-09:
The *levenberg-marquardt* solver of `fsolve` also hangs up on refprop limits. 

To try:
1. Use constrained solution function `fmincon` and set temperature limits as constraints as per refprop EOS constraints.  
2. Use polynomial approximations to refprop. 

LaTex file updated [fsolve_issues_Sept9_v2.pdf](https://github.com/nagimov/cyclotron-cryopanel/files/464767/fsolve_issues_Sept9_v2.pdf)

savreline, 2016-09-09:
No. 1 from above is done. 

Code uploaded that uses `fmincon`, subject to constraints to avoid running into refprop limits. Perhaps I like this better as it doesn't require us to modify refprop files!!

nagimov, 2016-09-14:
@savreline okay I got what you're doing wrong. You're solving the entire `F` matrix using `fsolve` (or any other solver) instead of finding the solution for each time step independently (i.e. solving it row by row). That complicates the problem significantly (think about solving that for t = 1000 hours, that'll be a zillion equation). We should only solve equations relevant to the current step at each iteration.

savreline, 2016-09-14:
@nagimov Yes of course!! I forgot that the time coordinate could be solved by steps, aggh! I was solving this like an `x` by `y` grid for the Laplace equation. Anyway I will work on the new code... I actually wrote code like this for the time problems before, will find it if need be. I will see if `fsolve` works better with new code and if not, then will try those other ideas we had. Thank you!

nagimov, 2016-09-15:
@savreline I've done something with python and [COBYLA](https://en.wikipedia.org/wiki/COBYLA) solver. It works pretty good but takes long time to converge (I'm sure it's mostly because of python being slow).
Here is the evolution of enthalpy (at constant pressure) along HX for both streams
[image]
Took me only ~4 minutes to get this for 10 slices and 20 timesteps!

savreline, 2016-09-16:
Very nice plot! Solution time looks promising!

savreline, 2016-09-26:
@nagimov What time step did you use? Mine is 0.01s. 
What were your tolerances? Mine are 1e-9.

nagimov, 2016-09-26:
@savreline 
```
tol = 1e-7
t_delta = 0.1  # integration time step, s
```

savreline, 2016-09-26:
Even through the solution converges very well (`fval` < 1e-6) I'm very suspicious of the plots. They reach steady-state way too fast and sometimes there are oscillations at the beginning. 

[image]
[image]
[image]

savreline, 2016-09-26:
Will study the code to see what could be wrong

savreline, 2016-09-29:
@nagimov We should close this issue as we are ahead of this step by now.

savreline, 2016-10-04:
Plot for Helium with phase-change occurring after a few time steps 

[image]
[image]

savreline, 2016-10-04:
@nagimov Better?

[image]

nagimov, 2016-10-07:
closing, done

closed 2016-10-07 by nagimov