nagimov/cyclotron-cryopanel #9: refprop investigations and approximations

savreline, 2016-09-09:
1. Find the function in refprop that determines quality of stream. 

*function not yet found - nothing in the help file or on the web*

2. Determine polynomial approximations to refprop functions.

savreline, 2016-09-09:
Turns out that in our range of temperatures:

H and U are 100% linear for nitrogen - we get an r2 value of 1.0000, so not issues fitting a line here.
H and U are almost linear for helium - we get an r2 value of .9984, so perhaps fitting a line is OK for now. We could also fit a second-order polynomial (parabola) and get an r2 value of .9999!!! 

[image]

nagimov, 2016-09-11:
@savreline 
- [ ] check the current version with the above approximations, post the results here
- [ ] wrap something around refprop and catch calls outside of allowed property regions to avoid exceptions
- [ ] check the formulations using simple case with ~2-3 slices and solving the system of equations manually (e.g. in excel)

Ultimately we want to have `fsolve` working fine with our equation at least for the easiest cases (low `UA`)

savreline, 2016-09-26:
Refrprop wrapper created, works well.

savreline, 2016-09-27:
Wrapper updated, reads error messages

nagimov, 2016-09-27:
+1 for a nice regexp parser :)

savreline, 2016-09-27:
@nagimov Matlab website says that all solvers like `fsolve` only *always* find a local minimum. They list procedures from global optimization toolbox on how to find global minimums.

savreline, 2016-09-27:
1. Tried global minimum search algorithms - they fail. 
They mainly fail because `fmincon` fails on this problem for some reason.

2. Tried to linearize refprop - r2 are good (almost a straight line for H-T and H-U at constant P).
The model solves way faster (~11s).
But the solution is still bad. Code uploaded under issue_09. 

Conclusion: need to check the model going further.

nagimov, 2016-10-07:
We're done with refprop and switched to coolprop for performance.

closed 2016-10-07 by nagimov