nagimov/cyclotron-cryopanel #10: Convert main code to P-H

savreline, 2016-09-26:


savreline, 2016-09-27:
<imgscr=https://cloud.githubusercontent.com/assets/21995425/18884507/87d2093e-849c-11e6-8f27-4a19678cc754.png width=150px> 
<imgscr=https://cloud.githubusercontent.com/assets/21995425/18884508/87e7d17e-849c-11e6-81e2-713cdf83fb76.png width=150px>

savreline, 2016-09-29:
Nice new plots! For 20 time steps and 10 slices. 

[image]
[image]

Updated code uploaded, issue_10 branch **restored**.

savreline, 2016-09-30:
Plots for 10 timesteps 

[image]
[image]
[image]

savreline, 2016-09-30:
Plots for 20 time steps 

[image] [image] [image]

savreline, 2016-09-30:
Plots for 20 time steps, with `HX_UA = 3000` (took the heat transfer coefficient lower, to prevent the plots from intersecting each other)

[image]
[image]
[image]

Still some issues for He, but better

nagimov, 2016-10-01:
@savreline let's look at heat load through iterations (plots which we've discussed today) before closing this one

savreline, 2016-10-03:
@nagimov The heat load looks OK. 

[image]
[image]

nagimov, 2016-10-03:
@savreline okay this one is just caused by reduced heat load at the end of iteration. Cold stream inlet temp is 100K and with a lower heat load its ability to warm up the cold stream within one slice is lower, so temperature is dropping near the inlet. Closing this issue.
Please submit the latest code as well.

closed 2016-10-03 by nagimov

savreline, 2016-10-03:
Latest code for this one was submitted