nagimov/cyclotron-cryopanel #11: Add pressure drop

savreline, 2016-09-29:
Remove assumption of constant pressure

savreline, 2016-09-29:
@nagimov Should we use something like `Delta P = K * m^2 / rho' for the pressure drop?

savreline, 2016-10-14:
@nagimov Plots for the pressure drop. As expected, He pressure (red) is decreasing and N2 pressure (blue) is increasing. Input He pressure is 1.25 atm and 1.75 atm for N2.

Not sure what is going on in the beginning, will study that.  

[image]
[image]

savreline, 2016-10-14:
68 second solution time on a `20x20x20` grid, with everything accounted for.

savreline, 2016-10-18:
[image]

[image]

savreline, 2016-10-19:
Low temperature solution

[image]
[image]
[image]
[image]

nagimov, 2016-10-31:
done, closing

closed 2016-10-31 by nagimov