nagimov/cyclotron-cryopanel #12: Add wall between HX streams

savreline, 2016-09-29:


savreline, 2016-09-30:
@nagimov Plots for the model with the wall. 20 time steps, 10 slices, take over 20 min to solve. Look OK I guess. I lower HX_UA a little bit, otherwise plots get too close and solution is not as good. 

[image]
[image]
[image]

savreline, 2016-09-30:
@nagimov With much higher `HX_UA`, 20 time steps, 10 slices. 

[image]

nagimov, 2016-10-01:
@savreline Let's not go crazy about calculating the temperature distribution along the wall as we discussed today, but simply calculate the heat flux through the wall assuming it has the thickness `b_w` and average conductance `k` calculated at average temperature `T` (e. g. calculated from `T_a` and `T_b` from previous time step). We can calculate the temperature distribution along the wall independently after calculating `T_a` and `T_b`, and gain some performance on the solver.

savreline, 2016-10-01:
@nagimov Agreed, +is this what you were looking for?

[image]

nagimov, 2016-10-01:
@savreline yes, looking good. Closing the issue

closed 2016-10-01 by nagimov