nagimov/cyclotron-cryopanel #18: Introduce limited heat conductance of the wall

nagimov, 2016-10-01:
We have decided to go with a simplified overall heat flux through the wall first, and use simplified (linear?) temperature distribution across the wall for calculations at each timestep

savreline, 2016-10-03:
Set up of energy balances for this case 

[Energy_balance.pdf](https://github.com/nagimov/cyclotron-cryopanel/files/507356/Energy_balance.pdf)

savreline, 2016-10-05:
Solution just barely converges for the case of 1 wall section... see plot
Doesn't converge for > 1 wall section. 

[image]

savreline, 2016-10-05:
LaTex file with equations is updated 

[Uploading EnergyBalance2.pdf…]()

savreline, 2016-10-07:
@nagimov There is actually a much easier way to tackle all of this. Since we have an explicit `dTw_dt` and all unknowns are `Tw` we can just use `ode45` for the `T` solver. And that shouldn't have those stupid convergence issues!!

nagimov, 2016-10-07:
@savreline how come it's explicit? We have not only `dTw/dt` but also `dTw/dz`, is that still feedable into ode45?

savreline, 2016-10-07:
@nagimov Yes! Because `ode45` integrates in time, not space. The `dTw/dz` in the wall can be treated like `dT/dx` was in the streams. And we used `ode45` for the streams before we had to use `du/dt` which is implicit in `T`.

savreline, 2016-10-07:
@nagimov Well 'dTw(i)/dt = (2*Tw(i)-Tw(i+1)-Tw(i-1))/m*cp(Tw(i))' ... there is an explicit function between 'dTw/dt' and 'Tw'.

nagimov, 2016-10-07:
@savreline sure! and we totally need a simple chart explaining which equations are solved each step (I'm lost already)

savreline, 2016-10-07:
@nagimov Will work on it!

savreline, 2016-10-08:
@nagimov Bam! If you use `HX_slices = 20` (before it was 10)

[image]
 works much better!

nagimov, 2016-10-08:
@savreline that is super-weird

savreline, 2016-10-11:
Some 3D plots 

[image]
[image]

savreline, 2016-10-11:
Fixed the code that uses only `fsolve`. The solution from that code is now the same as the `ode45` solution. So something is wrong with the equations, need to check.

savreline, 2016-10-12:
Plots for a 2-phase Helium stream 

[image]
[image]

savreline, 2016-10-14:
50 time steps

[image]

savreline, 2016-10-19:
Updated

[image]
[image]

savreline, 2016-10-19:
Even better for 50 time steps 

[image]
[image]
[image]

savreline, 2016-10-19:
Low temperature solution

[image]
[image]
[image]
[image]
[image]

nagimov, 2016-10-31:
done, closing

closed 2016-10-31 by nagimov