nagimov/cyclotron-cryopanel #20: Try working with CoolProp

savreline, 2016-10-03:
CoolProp installed as per instructions on their website; however, it doesn't work.

Will try again.

nagimov, 2016-10-03:
I'll work on coolprop.

savreline, 2016-10-03:
@nagimov Their sample code runs (which means CoolProp is installed). However, their sample code gives an error. 

[image]

nagimov, 2016-10-04:
[Wiki](https://github.com/nagimov/cyclotron-cryopanel/wiki/How-to-use-CoolProp-from-within-MATLAB) is updated to include the manual for two ways of using CoolProp: wrapper and c-like code.

savreline, 2016-10-04:
First version of wrappers that accept arrays are uploaded. Those wrappers are tested and do work, but do not catch many errors. To be improved on a daily basis.

nagimov, 2016-10-07:
I will rewrite `props_...` functions using c-like wrappers later (added to projects). This one is done and closed

closed 2016-10-07 by nagimov