nagimov/cyclotron-cryopanel #25: CoolProp: compare performances of c-like wrapper and "native" MATLAB wrapper

nagimov, 2016-10-10:


nagimov, 2016-10-10:
C-like wrapper is ~100 times faster in comparison with native MATLAB wrapper giving exact same results. Execution times in seconds (MATLAB wrapper, then c-like wrapper) and total absolute error:
```
>> tester_m
    6.8247
    0.0665
delta =
   4.9552e-11
```

**We are definitely replacing MATLAB wrapper with the one tested.**

closed 2016-10-10 by nagimov

savreline, 2016-10-20:
@nagimov  Difference between C-wrapper and Matlab wrapper 

For Helium
[image]

For Nitrogen
[image]