nagimov/cyclotron-cryopanel #28: Add radiative heat transfer

savreline, 2016-10-12:


savreline, 2016-10-12:
Some plots 

[image]
[image]
[image]

savreline, 2016-10-14:
@nagimov Added radiative heat transfer between `wall` and `ext`. So now we can see a bit of an increasing temperature distribution right in the middle of the wall as a function of time (green lines). Before there was only one solid line stuck at `150`. 

[image]

nagimov, 2016-10-16:
@savreline let's try to increase wall surface area to see the wall warming up both streams eventually

savreline, 2016-10-17:
@nagimov Here is if `As` is bumped up. Both streams heat up. If we increase it too much then weird things start to happen, will study those. 

[image]

savreline, 2016-10-19:
@nagimov  Plots for the model without the "tail" are much smoother now as well!!

[image]
[image]

savreline, 2016-10-19:
If we ram up radiation in the wall - to heat up streams 

[image]
[image]
[image]

savreline, 2016-10-19:
Solution for low T

[image]
[image]
[image]
[image]
[image]

nagimov, 2016-10-31:
done, closing

closed 2016-10-31 by nagimov