nagimov/cyclotron-cryopanel #29: memory leak

nagimov, 2016-10-12:
After the replacement of `props_...` function with c-like code, huge memory consumption is observed.
One run of the script (~few thousand function calls) consumes few GBs of RAM.
To be investigated

nagimov, 2016-10-12:
@savreline lets try to reload (unload and load back) coolprop library every now and then (say every 10th timestep or something like that). Do not do that every iteration, since that's a bit expensive.
So we now have `loadlibrary(libname,'CoolPropLib.h','includepath',path_to_include,'alias','coolprop');` loading the lib, to unload it just call `unloadlibrary libname`

nagimov, 2016-10-14:
it should be `unloadlibrary 'coolprop'`, just checked

savreline, 2016-10-14:
@nagimov Nine saw-tooth now. I have a variable `CP_dump = 10` in the code that we can adjust how often we wish to dump the library. I will set it to 20 right now, since I would like to dump only once during the iteration to save time. If you run the code on your computer, you can set that variable to 5 if you wish. 

[image]

nagimov, 2016-10-16:
solved, closing

closed 2016-10-16 by nagimov