nagimov/cyclotron-cryopanel #30: Dynamically compute U from fluid's properties

savreline, 2016-10-13:
@nagimov Those are the plots produced using Coolprop at  `1 atm` for `Pr, lambda, mu` as a function of `q`. Does that look reasonable? 

[image]

savreline, 2016-10-14:
Updated plots 

[image]

nagimov, 2016-10-16:
@savreline viscosity seems to be correct, here is the plot I found:
[image]
src: http://www.sciencedirect.com/science/article/pii/073519339500045Z (accessible from triumf LAN)

Also `lambda` and `cp` are expected to be linear. So `Prandtl` vs. `q` plot is okay.
Closing this one?

savreline, 2016-10-17:
@nagimov Yes, we should close

closed 2016-10-17 by savreline

savreline, 2016-10-19:
Low temperature solution 

[image]
[image]
[image]
[image]