nagimov/cyclotron-cryopanel #31: Temperature distribution in "wall's tail"

savreline, 2016-10-18:


savreline, 2016-10-18:
@nagimov This is really cool!!

Plots for the wall and wall tail profiles if `As` is everywhere `=1`
[image]
[image]

Plots for the wall and wall tail profiles if `As` in the wall tail is bumped up
[image]
[image]

savreline, 2016-10-19:
@nagimov The wall is fixed!!! 

[image]
[image]
[image]

savreline, 2016-10-19:
Low T solution - effect of delta_p??

[image]
[image]
[image]

nagimov, 2016-10-31:
done and verified, closing

closed 2016-10-31 by nagimov