nagimov/cyclotron-cryopanel #34: Link 4-stream model

savreline, 2016-10-28:
Link pairs of streams in a co-current and counter-current arrangements

savreline, 2016-10-28:
@nagimov Linked streams, counter-current solution 

[image]
[image]
[image]
[image]

savreline, 2016-10-28:
@nagimov Linked streams, with co-current solution 

[image]

nagimov, 2016-10-31:
done, linked

closed 2016-10-31 by nagimov